myhdl/myhdl #409: verify_convert() doesn't appear to work as expected

davekeeshan, 2023-02-03:
I can't imagine this is correct.  However when one is trying to verify the RTL sims of a testbench, they always seem to exit after the myhdl sim is run.

If the output of the myhdl sim is nothing, it assumes that there is nothing to verify, `_verify.py`:

```py
        if not flines:
            print("No MyHDL simulation output - nothing to verify", file=sys.stderr)
            return 1
```

It never gets to run the simulator version of the test.  As test finishing without an output is not unusual for a successful test, it is if there is a failure one might expect to see output.

Looking at the following code I am confused by what it is trying to attempt, so am not going to just smash fix it.

josyb, 2023-02-03:
delegating this to me then?
I am weary of touching those simulator things,.

davekeeshan, 2023-02-03:
@josyb ha ha, no I am happy enough to do it, I am just unclear what the testing is supposed to be doing, there there used to be a default statement at the end of every sum that guaranteed there was always at least one line of output?

davekeeshan, 2023-02-03:
Ok my initial analysis suggests that `StopSimulation` does not add any thing to the output that can be used to see if a test ran versus a tests that did not run.

It is possible to add to each test a print line like, `print("End of simulation")`, placed just before `StopSimultation`.  Now each simulator (myhdl, ghdl) print at least one line, and if it is the same line then both sims ran and they both ended the same way. It is a bit heavy.

It would be nice is the `StopSimulation` excpetion in the code could be used to determine whether or not a print line is also included in the code.  However to use a print line in vhdl you need to ad the line variable at the start of the process, so you need to know you are going to be printing.