myhdl/myhdl #411: Error: myhdl.AlwaysCombError: sensitivity list is empty

IamJezza, 2023-03-24:
Hi, 
I'm running the following python code:
```
from myhdl import block, instance, Signal, intbv, always_comb, delay

@block
def full_adder(a, b, cin, mysum, cout):
    @always_comb
    def code():
        mysum.next = a ^ b ^ cin
        cout.next = a and b or cin and (a or b)
    return code

@block
def twoBitAdder(A, B, mySum):
    w = Signal(intbv(0))
    f1 = full_adder(a=A[0], b=B[0], cin=0, mysum=mySum[0], cout=w)
    f2 = full_adder(a=A[1], b=B[1], cin=w, mysum=mySum[1], cout=None)
    return instance()

@block
def testBench1():
    A = Signal(intbv(2))
    B = Signal(intbv(2))
    mySum = Signal(intbv(2))
    FA = twoBitAdder(A, B, mySum)

    @instance
    def stimulus():
        for i in range(2):
            for j in range(2):
                A.next = i
                B.next = j
                yield delay(10)
                print(" A= %s | B= %s | S= %s " % (A, B, mySum))

    return FA, stimulus

tb1 = testBench1()
tb1.run_sim()
```

I'm getting the "myhdl.AlwaysCombError: sensitivity list is empty" error,
```
    raise AlwaysCombError(_error.EmptySensitivityList)
myhdl.AlwaysCombError: sensitivity list is empty

```
Can someone help me with this?
Thank you.

JanCoombs, 2023-03-24:
To get the simulation to run I had to eliminate the 'None'.  To achieve this I changed three more lines. Diff from original is now:
```'
jan:$ diff alwaysCombErr002_stripped.py alwaysCombErr003_stripped.py
7d6
< 
19c18
<     w = Signal(intbv(0)[1:])
---
>     w = [Signal(intbv(0)[1:]) for ii in range(2)] 
22,24c21,22
< 
<     f1 = full_adder(a=A(0), b=B(0), cin=0, mysum=mysum0, cout=w)
<     f2 = full_adder(a=A(1), b=B(1), cin=w, mysum=mysum1, cout=None)
---
>     f1 = full_adder(a=A(0), b=B(0), cin=0, mysum=mysum0, cout=w[0])
>     f2 = full_adder(a=A(1), b=B(1), cin=w[0], mysum=mysum1, cout=w[1])
```

josyb, 2023-03-24:
@JanCoombs I missed that `None` :(
As I prefer not to instantiate values that are written but not read (which will emit a conversion warning), I would write:
```python
@block
def full_adder(a, b, cin, mysum, cout):
    @always_comb
    def mksum():
        mysum.next = a ^ b ^ cin

    if cout is not None:
        @always_comb
         def mkcout():
            cout.next = (a & b) | (cin & (a | b))

    return instances()
```
I simplified my life by creating an OpenPort() which the discards the `.next` value in Simulation, and prefixes the conversion output line name with `--Openport` (`//OpenPort`  in Verilog) 
`f2 = full_adder(a=A(1), b=B(1), cin=w, mysum=mysum1, cout=OpenPort())`
You have to be careful however that such signals which can be set to `OpenPort` are not readback internally.

IamJezza, 2023-03-28:
Thank you @josyb

closed 2023-03-28 by IamJezza