myhdl/myhdl #414: different python versions

olisnr, 2023-04-04:
i had problems to get my code, that was running on one computer with python3.8, running on an other with python3.11. there were many strage messages, pointing deep in myHDLs verilog-converter. i needed long time to find out, that i dont need an old python version, but to update myHDL.

would it be esay to check the python version from myHDL and tell if the python version is suitable for the current myHDL version?

josyb, 2023-04-04:
Even if we added such a feature, that would not protect you from using an *outdated* MyHDL version (without this feature)  with a newer Python version.

olisnr, 2023-04-05:
yes, You are right. but it would give me a hint where to search: if i install myHDL via pip, i was thinking i have a good version. i tried different python-versions, and i had no idea what was needed. it should work from v3.6 and it was working on a 3.8, but not 3.10 and there is even a python 3.11 avaliable. 
the information, what python-versions are tested would have helped me. if its easy to implement i would print this information. the thing is: at the point where the new myHDL version is in the pip, it will work. 
it would also be ok, if the website says: the pip-version isnt new and should work at this and that versions

olisnr, 2023-04-06:
thats great, can we also have this information on "https://myhdl.org/start/installation.html#installation-using-pip"?

josyb, 2023-04-06:
That is another *section* :)

olisnr, 2023-04-11:
who is responsible for web-site changes?

olisnr, 2023-04-11:
there is also the broken link we could fix

josyb, 2023-04-11:
> who is responsible for web-site changes?

the community is ... I have merging rights, but that doesn't mean I do all the work :) @davekeeshan and @hgomersall  will help in reviewing PRs

> there is also the broken link we could fix

which one?

olisnr, 2023-04-11:
on https://github.com/myhdl/myhdl ther is an installation link:
pip install -e 'git+https://github.com/myhdl/myhdl#egg=myhdl
on the end is a ' missing. correct would it be:
pip install -e 'git+https://github.com/myhdl/myhdl#egg=myhdl' <--- this

olisnr, 2023-04-11:
thanks. i didn't saw it. there is an other thing thats wrong. the links of the page https://myhdl.org/start/installation.html goes to an old v0.9 verison on SourceForge instead of the actual GitHub version: 

To install MyHDL on your system, [download](http://sourceforge.net/project/showfiles.php?group_id=91207) the latest release. Untar and unzip the downloaded file:

josyb, 2023-04-11:
Like I said that is another section; perhaps @JanCoombs can find the time to work on that

bjohan, 2024-01-06:
I think i have a related problem where vhdl code generation works with python3.6 but not python3.11. 

I have the following component in a file called `component_configurable_trigger.py`:
```
from myhdl import *

@block
def configurable_trigger(dataIn, andMask, orMask, triggedAnd, triggedOr):

    @always_comb
    def trig_and():
        x = dataIn[0] and  andMask[0]
        for i in range(1, len(dataIn)):
            x = x and ((dataIn[i] and andMask[i])  or( not andMask[i]))
        triggedAnd.next = x

    @always_comb
    def trig_or():
        x = dataIn[0] or  orMask[0]
        for i in range(1, len(dataIn)):
            x = x or (dataIn[i] and orMask[i])
        triggedOr.next = x

    return trig_and, trig_or
```

I generate vhdl-code using a file named `generate_vhdl_configurable_trigger.py` that contains:
```
from myhdl import *

from component_configurable_trigger import configurable_trigger

def convert_configurable_trigger(hdl):
    
    dataIn = Signal(intbv(0)[4:])
    andMask = Signal(intbv(0)[4:])
    orMask = Signal(intbv(0)[4:])
    trigAnd = Signal(False);
    trigOr = Signal(False);
    
    configurable_trigger_inst = configurable_trigger(dataIn, andMask, orMask, trigAnd, trigOr)
    configurable_trigger_inst.convert(hdl=hdl);

convert_configurable_trigger(hdl='VHDL')
```

When I run the code generation using python3.6 `python3.6 generate_vhdl_configurable_trigger.py` I get the following output that I believe to be correct. 
```
-- File: configurable_trigger.vhd
-- Generated by MyHDL 0.11
-- Date: Sat Jan  6 15:10:03 2024

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_011.all;

entity configurable_trigger is
    port (
        dataIn: in unsigned(3 downto 0);
        andMask: in unsigned(3 downto 0);
        orMask: in unsigned(3 downto 0);
        triggedAnd: out std_logic;
        triggedOr: out std_logic
    );
end entity configurable_trigger;

architecture MyHDL of configurable_trigger is
[... 122 lines cut ...]
        andMask.next = 0b0101
        orMask.next = 0b1010

        for i in range(3):
            yield clk.posedge

        for b in range(16):
            dataIn.next = b
            yield clk.posedge

        for i in range(3):
            yield clk.posedge

        raise StopSimulation("Simulation stopped")

    return clkgen, configurable_trigger_inst, write, monitor

tb = test_configurable_trigger();
tb.config_sim(trace=True)
tb.run_sim(); 
```

I think the problem is when I'm indexing my intbv's such as: `x = dataIn[0] and  andMask[0]` but i have not found any work arounds.

Is this behavior correct? If so how would i change the code to work on newer python versions?

josyb, 2024-01-06:
@bjohan Can you please open a new issue for this, keeping things apart.
Thanks,
Josy