myhdl/myhdl #416: verilog and VHDL reset from @allways_seq() is strange

olisnr, 2023-04-24:
if i use an list of signals, its possible to set a few as logic, and the others as flip-flops. the generation of the reset if don't look what signal is set in the @allways_seq() procedure and sets he hole list of signals. i think only the FF should be reset.
and may be there should be an error when a signal has more then one source, i would see the problem even in the simulation.
[reset.zip](https://github.com/myhdl/myhdl/files/11313897/reset.zip)

josyb, 2023-04-25:
This a complicated one.
I find the example a bit contrived as it not exactly a real life thing; you are deliberately mixing registered and combinatorial signals in an array. 
Indeed MyHDL doesn't handle that well; one could have expected that only the actual signals used in the `@always_seq()` are reset and the other ones left untouched. I guess that could be remedied.
Note that Verilog can't handle that as it has to choose between a `wire` or a `reg`  to declare the `z` array object -> `wire signed [47:0] z [0:9-1];`, either will be wrong :( 
The SystemVerilog `logic` would be an outcome here. (I have an experimental branch for that)

olisnr, 2023-04-26:
Yes, mixing wires and registers its not really what verilog does. but even if You have a list of registers and use more then one @allways_seq() for a part of the registers, it would make resets with multiple sources.
for wire/register mixing, it could work, if we produce an array wires and an array registers and connect the registers to the wires where we need registers.
what do You think about multiple sources of wires? shouldent this be forbidden and generate an error? or is there a good reason to drive a digital signal from more then one sources (and i dont mean trhestate and the like).

olisnr, 2023-04-26:
i think it would be better, if the reset only sets the signals that are set in the @allways_seq()

josyb, 2023-04-26:
It looks to me that you are used to Verilog?

> if You have a list of registers and use more then one @allways_seq() for a part of the registers, it would make resets with multiple sources.

Just use a single `@always_seq` :)
In a module I only have a maximum of three processes: an `@always_comb`, an`@always_seq(clk.posedge, reset)` and (sometimes not present) `@always_seq(clk.posedge, reset=None)`

> what do You think about multiple sources of wires? shouldent this be forbidden and generate an error? 

Eventually the FPGA compiler will throw an error; but I understand you rather have MyHDL flag it first.

As I said, we could remedy the issue you have here, but it can be easily be worked around by separating the goat from the sheep (`wire` vs `reg` or *combinatorial* vs *clocked*) and declare separate variable sets.
IMO creating a list of Signals with different writing modes is not good practice.

olisnr, 2023-04-26:
im not used to verilog: i didnt write any verilog code my self. i did write some VHDL, but then i discovered myHDL. verilog is only the glue to the FPGA software, could also be VHDL but my FPGA software seems to like verilog more.

"separating the goat from the sheep" is what i did after i found that a "common stable" will result "miscarriage verilog" code ;)
the thing whit myHDL is: there is no problem using signals form a list in any way You want. as example. i made an register-pipeline, and in the end i wanted to change the input register to a wire, because the signal was already in sync. so i moved this signal form the seq to the comb block...

i do structure my blocks towards the functions and so i write normally many seq and comb functions in the same module. do You think that's also a bad habit?

and if so, wouldnt it be better to not produce verlog or VHDL code, if the design dont follow the rules that garantees correct code?

at least we should update the documentation: http://docs.myhdl.org/en/stable/manual/conversion.html like:
1. dont drive signal from a list from differenten processes
2. dont drive digital signals from different processes
3. dont write reset code if You use the @allways_seq() with a reset signal

olisnr, 2023-04-26:
_You can drive signals from a list from different processes; it is only that reset behavior which is causing a problem.
2.f you drive the same signal in different processes - MyHDL will produce a warning or error; it is only in your particular place that it doesn't - we have to look into this_

yes You can drive signals from different processes, but You will have this conversion-problems from _"mixing registered and combinatorial signals"_ in the verilog and VHDL output. so we should at least write: dont mix seq and and comb sources in the same list - if You want to generate code. may be in the **Conversion of lists of signals** section?

josyb, 2023-04-26:
I checked MyHDL sources and can't find an easy way to generate a warning/error.
It is clear that MyHDL considers a List of Signals as an *unit*; so you will see a reset assignment for every member.
So your proposal to add a warning in the documentation is the best we can do today.
One hiccup: until today I haven't done an update to the documentation package (-> [The MyHDL manual](https://docs.myhdl.org)) and will have to search for help there.

olisnr, 2023-05-09:
super. thanks!