myhdl/myhdl #420: is it possible to write a testbench.vcd from Cosimulation(...)?

olisnr, 2023-05-16:
i tried to change a python simulation in to a icarus-cosimulation. it seems to only work if i remove the **tb.config_sim(trace=True)** statement.
otherwise it says: _AttributeError: 'Cosimulation' object has no attribute 'name'_
is this intended?

olisnr, 2023-05-16:
may be the idea is that the verilog simulator writes the data. then there is may be a way to tell the mydesign.convert('verilog') function to write the tb_mydesign.v file to do this? or is it necessary to edit the testbench after generating it?

josyb, 2023-05-16:
I have run it once to test it but I don't remember what I actually did ...
I never use cosimulation and haven't felt the need for it. I have put my trust in MyHDL's simulation.  :)

josyb, 2023-05-17:
I don't know - I don't have one 
Most books I have on VHDL, Verilog and SystemVerilog describe the language and (try to) teach you how to use it, but don't teach how simulation works.
I searched a bit, as you made me curious :)
This looks a possible candidate: [Principles of Verifiable RTL Design: A functional coding style supporting verification processes in Verilog 2nd Edition](https://www.amazon.com/Principles-Verifiable-RTL-Design-verification/dp/0792373685) (I might buy it ...)
Perhaps this one: [Digital Logic Testing and Simulation](https://www.amazon.com/Digital-Logic-Testing-Simulation-Alexander/dp/0471439959)
Our BDFL, Jan Decaluwe : original author of MyHDL, has several posts on simulation; here is a link to one of them: [VHDL's crown jewel
](https://insights.sigasi.com/opinion/jan/vhdls-crown-jewel/)

closed 2023-10-04 by olisnr

changkaiyan, 2023-10-04:
您好，我将尽快回复您的邮件。如果您的邮件无需回复，则收到本回复表明我已收到您的邮件。请勿回复，谢谢合作。This is an automatic reply, confirming that your e-mail was received.Thank you.