myhdl/myhdl #425: Code generation working on python 3.6 but not 3.11

bjohan, 2024-01-06:
Summary

## System information
```
MyHDL Version: 0.11
Python Version: 3.6.9 ad 3.11.4
```
VHDL code generation works with python3.6 but not python3.11.

I have the following component in a file called component_configurable_trigger.py:

from myhdl import *

@block
def configurable_trigger(dataIn, andMask, orMask, triggedAnd, triggedOr):

    @always_comb
    def trig_and():
        x = dataIn[0] and  andMask[0]
        for i in range(1, len(dataIn)):
            x = x and ((dataIn[i] and andMask[i])  or( not andMask[i]))
        triggedAnd.next = x

    @always_comb
    def trig_or():
        x = dataIn[0] or  orMask[0]
        for i in range(1, len(dataIn)):
            x = x or (dataIn[i] and orMask[i])
        triggedOr.next = x

    return trig_and, trig_or

I generate vhdl-code using a file named generate_vhdl_configurable_trigger.py that contains:

from myhdl import *

from component_configurable_trigger import configurable_trigger

def convert_configurable_trigger(hdl):
    
    dataIn = Signal(intbv(0)[4:])
    andMask = Signal(intbv(0)[4:])
    orMask = Signal(intbv(0)[4:])
    trigAnd = Signal(False);
    trigOr = Signal(False);
    
    configurable_trigger_inst = configurable_trigger(dataIn, andMask, orMask, trigAnd, trigOr)
    configurable_trigger_inst.convert(hdl=hdl);

convert_configurable_trigger(hdl='VHDL')

When I run the code generation using python3.6 python3.6 generate_vhdl_configurable_trigger.py I get the following output that I believe to be correct.

-- File: configurable_trigger.vhd
-- Generated by MyHDL 0.11
-- Date: Sat Jan  6 15:10:03 2024

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_011.all;

entity configurable_trigger is
    port (
        dataIn: in unsigned(3 downto 0);
        andMask: in unsigned(3 downto 0);
        orMask: in unsigned(3 downto 0);
        triggedAnd: out std_logic;
        triggedOr: out std_logic
    );
end entity configurable_trigger;

architecture MyHDL of configurable_trigger is

begin

CONFIGURABLE_TRIGGER_TRIG_AND: process (andMask, dataIn) is
    variable x: std_logic;
begin
    x := stdl(bool(dataIn(0)) and bool(andMask(0)));
    for i in 1 to 4-1 loop
        x := stdl(bool(x) and ((bool(dataIn(i)) and bool(andMask(i))) or (not bool(andMask(i)))));
    end loop;
    triggedAnd <= x;
end process CONFIGURABLE_TRIGGER_TRIG_AND;

CONFIGURABLE_TRIGGER_TRIG_OR: process (orMask, dataIn) is
    variable x: std_logic;
begin
    x := stdl(bool(dataIn(0)) or bool(orMask(0)));
    for i in 1 to 4-1 loop
        x := stdl(bool(x) or (bool(dataIn(i)) and bool(orMask(i))));
    end loop;
    triggedOr <= x;
end process CONFIGURABLE_TRIGGER_TRIG_OR;

end architecture MyHDL;

However using python 3.11 python3.11 generate_vhdl_configurable_trigger.py I get an AttributeError.

Traceback (most recent call last):
  File "/home/bjohan/git_wd/xilinx_ml505_makefile_template/myhdl/generate_vhdl_configurable_trigger.py", line 18, in <module>
    convert_configurable_trigger(hdl='VHDL')
  File "/home/bjohan/git_wd/xilinx_ml505_makefile_template/myhdl/generate_vhdl_configurable_trigger.py", line 15, in convert_configurable_trigger
    configurable_trigger_inst.convert(hdl=hdl);
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/_block.py", line 342, in convert
    return converter(self)
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_toVHDL.py", line 204, in __call__
    genlist = _analyzeGens(arglist, h.absnames)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 170, in _analyzeGens
    v.visit(tree)
  File "/usr/lib/python3.11/ast.py", line 418, in visit
    return visitor(node)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 1119, in visit_Module
    _AnalyzeBlockVisitor.visit_Module(self, node)
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 1070, in visit_Module
    self.generic_visit(node)
  File "/usr/lib/python3.11/ast.py", line 426, in generic_visit
    self.visit(item)
  File "/usr/lib/python3.11/ast.py", line 418, in visit
    return visitor(node)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 1099, in visit_FunctionDef
    self.visit(n)
  File "/usr/lib/python3.11/ast.py", line 418, in visit
    return visitor(node)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 528, in visit_Assign
    self.visit(value)
  File "/usr/lib/python3.11/ast.py", line 418, in visit
    return visitor(node)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 446, in visit_BoolOp
    self.visit(n)
  File "/usr/lib/python3.11/ast.py", line 418, in visit
    return visitor(node)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 962, in visit_Subscript
    self.accessIndex(node)
  File "/usr/lib/python3/dist-packages/myhdl/conversion/_analyze.py", line 986, in accessIndex
    self.visit(node.slice.value)
  File "/usr/lib/python3.11/ast.py", line 418, in visit
    return visitor(node)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ast.py", line 422, in generic_visit
    for field, value in iter_fields(node):
  File "/usr/lib/python3.11/ast.py", line 260, in iter_fields
    for field in node._fields:
                 ^^^^^^^^^^^^
AttributeError: 'int' object has no attribute '_fields'

I also have a small test bench to exercise the component that works with both python3.6 and python3.11.

from myhdl import *

from component_configurable_trigger import configurable_trigger
@block
def test_configurable_trigger():
    clk = Signal(False)
    reset = ResetSignal(0, active=1, isasync=False)
    
    dataIn = Signal(modbv(0)[4:])
    andMask = Signal(modbv(0)[4:])
    orMask = Signal(modbv(0)[4:])
    trigAnd = Signal(False)
    trigOr = Signal(False)

    @always(delay(10))
    def clkgen():
        if clk:
            clk.next = 0
        else:
            clk.next = 1

    configurable_trigger_inst = configurable_trigger(dataIn, andMask, orMask, trigAnd, trigOr)

    @instance
    def monitor():
        for nclk in range(100):
            yield clk.posedge
        raise StopSimulation("Too many clk cycles elapsed, increase nclk if needed")

    @instance
    def write():
        reset.next = 1
        for i in range(3):
            yield clk.posedge

        reset.next = 0
        andMask.next = 0b0101
        orMask.next = 0b1010

        for i in range(3):
            yield clk.posedge

        for b in range(16):
            dataIn.next = b
            yield clk.posedge

        for i in range(3):
            yield clk.posedge

        raise StopSimulation("Simulation stopped")

    return clkgen, configurable_trigger_inst, write, monitor

tb = test_configurable_trigger();
tb.config_sim(trace=True)
tb.run_sim(); 

I think the problem is when I'm indexing my intbv's such as: x = dataIn[0] and  andMask[0] but i have not found any work arounds.

Is this behavior correct? If so how would i change the code to work on newer python versions?

josyb, 2024-01-06:
@bjohan 
You are not using the latest version of MyHDL
I guess you have installed MyHDL from PyPi, but unfortunately that version is lagging quite a bit:
[image]

I just tested your code against the current repo master and it compiles fine:
```VHDL
-- File: configurable_trigger.vhd
-- Generated by MyHDL 0.11.43
-- Date: Sat Jan  6 15:49:03 2024

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_011.all;

entity configurable_trigger is
	port(
		dataIn     : in  unsigned(3 downto 0);
		andMask    : in  unsigned(3 downto 0);
		orMask     : in  unsigned(3 downto 0);
		triggedAnd : out std_logic;
		triggedOr  : out std_logic
	);
end entity configurable_trigger;

architecture MyHDL of configurable_trigger is

begin

	CONFIGURABLE_TRIGGER_TRIG_AND : process(andMask, dataIn) is
		variable x : std_logic;
	begin
		x          := stdl(bool(dataIn(0)) and bool(andMask(0)));
		for i in 1 to 4 - 1 loop
			x := stdl(bool(x) and ((bool(dataIn(i)) and bool(andMask(i))) or (not bool(andMask(i)))));
		end loop;
		triggedAnd <= x;
	end process CONFIGURABLE_TRIGGER_TRIG_AND;

	CONFIGURABLE_TRIGGER_TRIG_OR : process(orMask, dataIn) is
		variable x : std_logic;
	begin
		x         := stdl(bool(dataIn(0)) or bool(orMask(0)));
		for i in 1 to 4 - 1 loop
			x := stdl(bool(x) or (bool(dataIn(i)) and bool(orMask(i))));
		end loop;
		triggedOr <= x;
	end process CONFIGURABLE_TRIGGER_TRIG_OR;

end architecture MyHDL;
```
Best regards,
Josy

bjohan, 2024-01-06:
Thank you very much! This completely solved it.

Best regards
Johan

closed 2024-03-14 by josyb