myhdl/myhdl #429: Bit indexing may cause deprecation warnings

zhijieshi, 2024-02-17:
Summary

## System information
```
MyHDL Version: 0.11.45
Python Version: 3.12.2
```

The example below gives Deprecation Warnings. It looks like bit indexing, e.g., `s[0]`, returns Bool type. 

```
DeprecationWarning: Bitwise inversion '~' on bool is deprecated.
```

```
from myhdl import block, intbv, instance, delay, Signal, StopSimulation, always_comb

@block
def test_block():

    # create signals. 
    s = Signal(intbv(0)[2:])
    output = Signal(intbv(0)[1:])

    @always_comb
    def xor_logic():
        output.next = ((~s[1] & s[0]) | (s[1] & ~s[0])) & 1

    @instance
    def stimulus():
        s.next = 3
        yield delay(10)
        print("output is ", output)

        s.next = 2
        yield delay(10)
        print("output is ", output)

        raise StopSimulation()
    return xor_logic, stimulus

tb = test_block()
tb.run_sim()
```

josyb, 2024-02-17:
No offence, but it is good practice to leave the 'severity level' of an issue to the maintainers.

josyb, 2024-02-17:
I see that you are using Python 3.12, but MyHDL is not yet checked to that version.
I installed 3.12 and yes the deprecation warning shows. But 3.11 is fine.

zhijieshi, 2024-02-18:
Sorry. I did not see better options. Will be more careful next time.

josyb, 2024-02-18:
There was/is nothing wrong with your code.
Until the Python lot decide to make this (silly IMO) change.
I complained about that here: https://github.com/python/cpython/pull/103487#issuecomment-1950294605
Not sure I will get a reply?

timhoffm, 2024-02-18:
Please note that the bit inversion `~` of a bool does not behave as a logical negation, in particular `bool(~True) is True`. If you have used `~` meaning logical negation, you code is broken. The deprecation is there to make you aware of the issue. Please use the logical negation operator `not` instead.

zhijieshi, 2024-02-18:
Here is an example where I think bit-wise operations are more concise. 

```Python

# use not for negation
z.next =  ((not s[1]) & (not s[0]) & a) | 
	  ((not s[1]) &      s[0]  & b) | 
	  (     s[1]  & (not s[0]) & c) | 
	  (     s[1]  &      s[0]  & d)

# use bitwise operations
z.next = ((~s[1] & ~s[0] & a) | 
	  (~s[1] &  s[0] & b) | 
	  ( s[1] & ~s[0] & c) | 
	  ( s[1] &  s[0] & d)) & 1
```

timhoffm, 2024-02-18:
You can do that, but I really would recommend to convert the inputs in such a case, e.g. `s0 = int(s[0])`. The code only works because of the trailing `& 1`, which is non-trivial to understand and easy to forget.

zhijieshi, 2024-02-19:
Yes. Converting single indexed bits to int first is an option. Or save the negated values (with `not`) in a variable first. 

The final masking `& 1` is still needed (sometimes) if `~` is used. I can imagine people make the same mistakes even if they are forced to convert Bool to int.

josyb, 2024-02-19:
We are now discussing this in two threads.
I suggest we continue in the other Python thread, as that is where the issue stems from.
At least I will only reply over there.