myhdl/myhdl #434: Inconsistent behavior  of assignment of Signal(intbv()[w:]) to Signal(bool())

josyb, 2024-06-21:
## System information
```
MyHDL Version: 0.12: experimental work on new converter
Python Version: 3.10+
```
I am working on a new converter approach to accommodate future target languages (first in line: SystemVerilog).
It can be seen as a two-stage approach: a common analysis followed by target writers.
Verilog is working fine. 
For VHDL I felt a rewrite is due: the current approach generates to many `resize(v, W)` which are not always well digested vy the Vendor tools (I ran into a particular problem with Vivado).
This obviously needs meticulous testing.

I stumbled into an inconsistency in `test\general\test_binops.py`.
In the HDL code we have this continuous assignment: ` x.next = a` with `x = Signal(bool(0))` and `a = Signal(intbv(0)[2:])`
1) Simulation will accept an `int` (`bool` is implicit) or an `intbv`  but will fail if the value of a is not in [0, 1]
2) Conversion to Verilog produces `assign x = a;` (I have to find out whether this is OK, but the Sigasi linter stays quiet ...)
 Conversion to VHDL produces `x <= a(0)`: this is inconsistent with 1) as e.g. a value of 3 will work fine in VHDL simulation but will fail in MyHDL's simulation.

My thoughts:
1) Simulation should only accept values in `[True, False, 0, 1, Signal(bool())]`
2) conversion should fail also if `a` is not in  `[True, False, 0, 1, Signal(bool())]`
This will result in *explicit* Python code, which will generate *equivalent* functionality in Verilog and VHDL (and SystemVerilog ...)

JanCoombs, 2024-07-14:
Agreed. How can I help?

Before starting to use effective test frameworks I used the Verilog and VHDL conversions to gain confidence. The VHDL conversions failed more frequently than the Verilog ones, but I was unable to discover why that was.  Both fixing the current and your development will likely fix this.

josyb, 2024-07-14:
Hi Jan,

I wanted your insights on this issue - and whether you can agree to how I would solve this. The only thing bothering me is that this change may/will break code; something I am desperately to avoid.
I can run all VHDL and Verilog tests without problems - even from within the Eclipse IDE I am still using. One  of these days I am forced to switch to Visual Studio Code as [Sigasi](https://sigasi.com) are going to drop Eclipse support :(