myhdl/myhdl #437: Tests are not passing?

Kreijstal, 2024-08-04:
I simply cloned and installed..

```
kreijstal@kreijstalnuc:~/git/workspace/cosim$ cd myhdl/test/core
py.test
sh: cd: myhdl/test/core: Datei oder Verzeichnis nicht gefunden
============================================= test session starts =============================================
platform linux -- Python 3.12.4, pytest-7.0.1, pluggy-1.2.0
rootdir: /home/kreijstal/git/workspace/cosim
plugins: timeout-2.1.0, typeguard-4.3.0, anyio-4.4.0
collected 0 items / 1 error

=================================================== ERRORS ====================================================
________________________________________ ERROR collecting test session ________________________________________
../../../.local/lib/python3.12/site-packages/_pytest/config/__init__.py:1492: in getoption
    val = getattr(self.option, name)
E   AttributeError: 'Namespace' object has no attribute 'sim'

The above exception was the direct cause of the following exception:
../../../.local/lib/python3.12/site-packages/pluggy/_manager.py:158: in register
    hook._maybe_apply_history(hookimpl)
../../../.local/lib/python3.12/site-packages/pluggy/_hooks.py:496: in _maybe_apply_history
    res = self._hookexec(self.name, [method], kwargs, False)
../../../.local/lib/python3.12/site-packages/pluggy/_manager.py:112: in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
myhdl/myhdl/test/conftest.py:19: in pytest_configure
    sim = config.getoption('sim')
../../../.local/lib/python3.12/site-packages/_pytest/config/__init__.py:1503: in getoption
    raise ValueError(f"no option named {name!r}") from e
E   ValueError: no option named 'sim'
=========================================== short test summary info ===========================================
ERROR  - ValueError: no option named 'sim'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
============================================== 1 error in 0.23s ===============================================
kreijstal@kreijstalnuc:~/git/workspace/cosim$
```

josyb, 2024-08-05:
Continuing here from discussion #436 
I assume you cloned the MyHDL repo from GitHub
From the `ast.Str` errors I deduce that you did not clone the latest MyHDL version, which should be 0.11.49.
Can you check MyHDL's `__version__`?

Kreijstal, 2024-08-05:
I cloned master, is master not the latest?

Kreijstal, 2024-08-05:
seems installing myhdl from pip was a mistake, always install from git

closed 2024-08-05 by Kreijstal

josyb, 2024-08-05:
> seems installing myhdl from pip was a mistake, always install from git

Perhaps it is an issue with your Linux distro. I did install from PyPi using pip without issue, but it was on a Windows machine :)