myhdl/myhdl #438: Error when converting a a tri state signal to VHDL

tobygomersall, 2024-08-06:
I am writing a block to instantiate a tri-state buffer for every bit in an n bit signal. This is intended for use at the top level, to help tidy up the Vivado block diagram (I can instantiate one of these blocks rather than n slicers, n tri-state buffers and a concat block).

When I try to convert to VHDL I get an error. When converting to Verilog, it seems to work correctly.

The conversion code fails on line 477 of `myhdl/mydhl/conversion/_toVHDL.py` (note: in the latest version of myhdl it has moved to line 484). This code snippet starts on line 475 of my version (line 482 of the latest):

```
if isinstance(sig_vhdl_obj, vhd_std_logic):
    # Single bit
    val_str = " := '%s'" % int(s._init)
```

If a signal on the interface is a `TristateSignal` then `s._init` is `None` and `int()` raises an error.

As a test, I managed to solve this by modifying the above code to:

```
if s._init is None:
    val_str = ''
elif isinstance(sig_vhdl_obj, vhd_std_logic):
    # Single bit
    val_str = " := '%s'" % int(s._init)
```

## System information
```
MyHDL Version: 0.11 (commit: 0477c7e042aee72fbd2afb722e68f6dee61c102c)
Python Version: 3.10.12
```

josyb, 2024-08-07:
Hi Toby,

Can you add a MRE (minimal reproducible example) or at least a snippet of your code? I will (try to) add it to the test-suite.
About your workaround: that is inside the `_writeSigDecls()` function. Does that mean that you have internal tri-state signals? I would have expected to only see them in the top-level ports ...
So far I have never used tristate signals in MyHDL code; in stead using the 3-signal method instead: `sigout`, `sigoutdisable` for output and `sig` itself for input, and then in the top-level VHDL file connecting them to a Vendor primitive or equivalent.

tobygomersall, 2024-08-07:
Hi Josy,

I am using myhdl to instantiate the Xilinx IOBUF primitives. The myhdl is also slicing the write signal and concatenating the read signals before connecting them to the IOBUFs. This block is intended to be converted in isolation and dropped on to the Vivado block diagram.

For testing, I create a DUT which instantiates two of these blocks with the bi-directional signals connected between the two. It is then easy to test the behaviour through the two IOBUFs. We need to do this because our testing framework cannot handle conversion tests when there are bi-directional signals on the interface of the DUT.

So yes, there are internal tri-state signals when we simulate.

[Here](https://gist.github.com/tobygomersall/d3adf72bf20de9a5449f567513a6bc92) is minimal reproducible example. Place both files in a test directory and run `python3 minimal_reproducible_example.py`.

josyb, 2024-08-07:
Hi Toby,
I was able to run your example with a small modification to `_toVHDL` avoiding initialisation of *any* `TristateSignal`:
```python
            val_str = ''
            if toVHDL.initial_values and not isinstance(s, _TristateSignal):
                if isinstance(sig_vhdl_obj, vhd_std_logic):
. . .

```
(at line 478)
But I am not yet convinced: the Verilog conversion has an empty module definition and the VHDL conversion has an empty port list. As soon as the `TristateSignal` is mentioned in the top-level function definition it will be no longer present in the internal signals list (written by `_writeSigDecls()`). There may be a point left in running a V* simulation perhaps ...

josyb, 2024-08-07:
Toby,
Putting in the change doesn't hurt, so I merged the solution. You can now update your local MyHDL package/tree.
The `toVerilog.py` code kind of did what we are doing now, relying on the `wire` status of the TristateSignal

Best regards,
Josy

tobygomersall, 2024-08-08:
Thanks Josy. In your previous post you mentioned an empty module definition (Verilog conversion) and and empty port list (VHDL conversion). Did you solve these issues?

josyb, 2024-08-08:
Toby,
The empty entity/module are a result of your code; you effectively convert a function without arguments; and that is the reason why you have an internal tristate signal.
While studying your case I modified your code to accept arguments and then the tristate signal is converted into an `inout' port:
[toby.zip](https://github.com/user-attachments/files/16550957/toby.zip)