myhdl/myhdl #441: New release or new version to pypi?

TurnOffNOD, 2024-09-02:
Since pypi version is quite behind the current development status, will there be new release in the near future? Or new dev version upload to pypi without new release?

josyb, 2024-09-02:
Actyally this is no longer the case; we update the PyPi after every PR merge. Currently we are at 0.11.49.
Unfortunately I haven't  updated MyHDL's *readme.md* to relefect that (yet, will try to do so at next merge).

TurnOffNOD, 2024-09-03:
> Actyally this is no longer the case; we update the PyPi after every PR merge. Currently we are at 0.11.49. Unfortunately I haven't updated MyHDL's _readme.md_ to relefect that (yet, will try to do so at next merge).

Oh, I see. I installed from pip as a result.

closed 2024-09-03 by TurnOffNOD