myhdl/myhdl #442: Type and input/output checks within block

sideprojectslab, 2024-09-26:
Hello, I have 10 years of professional experience in VHDL and I am investigating myHDL for my firm as an alternative to VHDL. myHDL looks really interesting, well thought out and polished, but since the VHDL design patterns are so etched in my mind I am having some trouble understanding how typing and port direction works.

Namely, it appears to me that a "block" cannot impose the type nor the direction of its ports in myHDL, which are instead determined by how the module is used. This is in contrast with VHDL where every instance of the same entity behaves identical.

One example of the issue is the following, I have a simple Blink block like this:
'''
from myhdl import *

class BlinkIO:
	def __init__(self):
		self.enable = Signal(bool(0))
		self.blink  = Signal(bool(0))

@block
def Blink(clk:bool, rst:bool, io:BlinkIO, period):

	count = Signal(modbv(0)[20:])

	@always(clk.posedge)
	def logic():
		if io.enable:
			if count == period:
				count.next = 0
				io.blink.next = not io.blink
			else:
				count.next = count + 1

		if rst:
			count.next = 0

	return logic
'''

and I am testing this module with this simple harness:
'''
from myhdl import *
from blink import *

@block
def test():

	clk = Signal(bool(0))
	rst = Signal(bool(1))
	io = BlinkIO()

	@always(delay(10))
	def clkgen():
		clk.next = not clk

	dut = Blink(clk, rst, io, 100)

	@instance
	def stimulus():

		yield clk.posedge
		rst.next = 0
		io.enable.next = 1

		for i in range(10):
			yield io.blink.posedge

		raise StopSimulation

	return clkgen, dut, stimulus

def run():
	i = test()
	i.config_sim(trace=True, tracebackup=False)
	i.run_sim(quiet=1)

if __name__ == "__main__":
	run()
'''

So far so good, it tests as expected and I can see from the generated VHDL that, the io.enable port is correctly interpreted as an input std_logic.

However if I change the Blink block by adding an assignment to io.enable, like this:
'''
from myhdl import *

class BlinkIO:
	def __init__(self):
		self.enable = Signal(bool(0))
		self.blink  = Signal(bool(0))

@block
def Blink(clk:bool, rst:bool, io:BlinkIO, period):

	count = Signal(modbv(0)[20:])

	@always(clk.posedge)
	def logic():
		if io.enable:
			if count == period:
				count.next = 0
				io.blink.next = not io.blink
				io.enable.next = 0
			else:
				count.next = count + 1

		if rst:
			count.next = 0

	return logic
'''

Now if I generate Blink on its own io.enable is converted as an output std_logic. In addition to this, within the test the io.enable signal is just driven high first by the stimulus, and then low by the dut (blink) instance, without generating a conflict.

Something similar happens the other way around, if I change my test like this:
'''
from myhdl import *
from blink import *

@block
def test():

	clk = Signal(bool(0))
	rst = Signal(bool(1))
	io = BlinkIO()

	fakerst = Signal(modbv(0)[10:])

	@always(delay(10))
	def clkgen():
		clk.next = not clk

	dut = Blink(clk, rst, io, 100)
	fakedut = Blink(clk, fakerst, io, 50)

	@instance
	def stimulus():

		yield clk.posedge
		rst.next = 0
		io.enable.next = 1

		for i in range(10):
			yield io.blink.posedge

		raise StopSimulation

	return clkgen, dut, fakedut, stimulus

def run():
	i = test()
	i.config_sim(trace=True, tracebackup=False)
	i.run_sim(quiet=1)

if __name__ == "__main__":
	run()
'''

Now I have two instances of the same block that apparently test just fine (the test even completes and passes), but their reset signal is a different type (also different in the converted VHDL), and while in this case there are no serious implications, I would assume this is not a desirable situation. The way I will think about it is that the port type itself could be seen as a parameter itself, which can actually be a powerful feature, but I would like to have a way to prevent this from happening when needed.

In addition, despite the fact that both dut and fakedut drive the same io.blink signal, no conflict is detected in the test.

So my question is, is there a way to get around these issues and make the behavior of a block more consistent, at the very least with asserts that check for type and direction of ports? And what is the approach to detecting conflicts when multiple instances drive the same signals?

josyb, 2024-09-26:
Hi,

Welcome to MyHDL!

There is a substantial difference in simulation and conversion; when simulating you are running the actual Python code just as you would run any other Python script as e.g. calculating the mean and deviation of a population's ages; in conversion your HDL code is compiled into the AST (Abstract Syntax Tree) which is then parsed and converted into VHDL (or Verilog).

In order of your post:
> Now if I generate Blink on its own io.enable is converted as an output std_logic. In addition to this, within the test the io.enable signal is just driven high first by the stimulus, and then low by the dut (blink) instance, without generating a conflict.

Yes, in conversion MyHDL infers the direction of the signal in function of its usage, as a consequence `io.enable` will become an output and as the initial value in hardware (FPGA)  will be zero the counter will not even start. There is a design pattern to avoid that, we will get to that later.

> Now I have two instances of the same block that apparently test just fine (the test even completes and passes), but their reset signal is a different type (also different in the converted VHDL), and [...]
...
In addition, despite the fact that both dut and fakedut drive the same io.blink signal, no conflict is detected in the test.

As I mentioned above simulation is simply running the Python code you have written, and as such a global variable, which `io.enable` and `io.blink` are can be altered by any function. 
If you look at the simulation waveform you can see that the output is not what you expect:
[image]
The `io.blink` output looks a bit random ... (I set both periods to a prime number, 97 and 37 to emphasize it)

Both versions, one with a `Signal(bool(0))` as reset and the other with a `Signal(modbv(0)[10:]) as reset, will convert to VHDL without issue.
If however you use multiple instances in a top level module, the conversion will fail:
```python
@block
def doubleblink(clk, reset, io):
    blinkright = blink(clk, reset, io, 97)
    blinkleft = blink(clk, reset, io, 37)

    return instances()

dfc3 = doubleblink(clk, rst, io,)
dfc3.convert(hdl='VHDL')
```
->

```
  File "C:\myhdlsupport\myhdl\myhdl\conversion\_misc.py", line 144, in raiseError
    raise ConversionError(kind, msg, info)
myhdl.ConversionError: in file C:\myhdlsupport\workspace-MyHDL-support\issues\sideprojectslab\blinky.py, line 21:
    Signal has multiple drivers: io_blink
```

Note that `doubleblink()` will simulate fine, regardless.

Coming back to the *design pattern* I mentioned; this is how I would (re-)write your module:
```python
from myhdl import Signal, always_seq, intbv, modbv, block, Constant, instances

class BlinkIO:

    def __init__(self):
        self.enable = Signal(bool(1))
        self.blink = Signal(bool(0))

@block
def myblink(clk, reset, io, period):
    count = Signal(modbv(0)[20:])

    @always_seq(clk.posedge, reset=reset)
    def synch():

        if io.enable:
            if count == period:
                count.next = 0
                io.blink.next = not io.blink
                # stop counting (and blinking ...)
                io.enable.next = 0

            else:
                count.next = count + 1
[... 118 lines cut ...]

@block
def blink(clk:Signal(bool), rst:Signal(bool), io:BlinkIO, period: Signal(intbv) | Constant(intbv) | int):

    assert isinstance(clk, SignalType) and(isinstance(clk.val, bool))
    assert isinstance(rst, SignalType) and(isinstance(rst.val, bool))

    count = Signal(modbv(0)[20:])

    @always(clk.posedge)
    def logic():
        . . .
```
These asserts will not convert to VHDL, as they are outside the `@always` generators
Note that `Signal()` is a function returning a class, so you must use `SignalType` to check this
I see that you you use type hinting; I changed these but I have no clue whether this will work with a linter - as I haven't tried this (yet)

> And what is the approach to detecting conflicts when multiple instances drive the same signals?

Conversion will pick this up; simulation would need extra code to detect this, it may be less trivial than we think.

Don't hesitate to ask more questions. But it would be preferable to pose these on either [Gitter](https://app.gitter.im/?updated=1.11.73#/room/#myhdl_myhdl:gitter.im/$bpXPHwD7n78yzT_m2JYyV3AetU_pYedkspBaA8tSRiM) or (perhaps better suited to *larger* questions like the one here) on [Discourse](https://discourse.myhdl.org)

Best regards,
Josy

sideprojectslab, 2024-09-26:
Hello Josy, thank you for your quick and thorough answer. I will switch to "discourse" for my future questions, but I guess it would make sense to continue this specific discussion here.

First of all I should clarify that that assignment to the io.enable pin had the sole purpose of checking how the tool would behave in the case of a common mistake such as assigning a value to an input port in VHDL, so it wasn't my intention to make it work in any meaningful way, but rather see at which stage of the implementation-simulation/testing-synthesis process would I be able to catch this sort of mistakes. So it is absolutely true that the module behaves erratically and any serious testing would spot the problem regardless of the conflict detection, but again my purpose here is to get a feeling of where we stand compared to a VHDL workflow.

So from your answer I believe I can gather the following points (please correct me if I am wrong):
1) It makes sense that converting a fully constrained design that contains multiple drivers catches the problem, and I was able to assess that shortly after I posted the question
2) It makes sense that port direction is inferred by the use it is made of that port within a block. I can (learn to) live with that.
3) I can assert on the port type in python to force a certain type (perhaps even width?) on ports
4) Testing in python does not "resolve" std_logic signal in the classical sense, but rather runs the description sequentially, let's say

My follow-up questions are:
1) would co-simulating with icarus detect multiple drivers? my assumption would be yes, and I'll soon put it to the test
2) for the purpose of asserting on assignments to input ports, I see that the TristateSignal type has the desirable feature of not having a "next" attribute, so I imagine that asserting on that type in a @block would ensure that one can't assign a value to an input port within an @always. Can you think of a clean design pattern around this?

I am asking this because I am really interested in the python simulation step, as a mean for fast iterations before switching to Icarus, so I am trying to gauge how much trust in the design correctness can we reach with this stage alone. Not saying it is too little, just trying to get a feeling.

Thank you again for your help.

P.S. When I try to login to discourse with github or google I get an error saying "invalid redirect URI"

josyb, 2024-09-26:
Hi,

Continuing here is perfectly fine.

> 1. It makes sense that converting a fully constrained design that contains multiple drivers catches the problem, and I was able to assess that shortly after I posted the question

Yes, conversion will catch multiple drivers - alas simulation won't

> 2. It makes sense that port direction is inferred by the use it is made of that port within a block. I can (learn to) live with that.

Read 3.

> 3. I can assert on the port type in python to force a certain type (perhaps even width?) on ports

Outside the `@always` generators you have the full power of Python at your disposal.
Inside the generators you have to be careful if you want them to convert. As a result I only (most of the time) use `@instance` in  my test benches.

I use a different MyHDL paradigm which I started many years ago,  [you can see the initial thoughts here](https://gist.github.com/josyb/d119a2f3f12dcc28533ac612e30576af)
It defines logic inside a class rather than as a function. In the (mandatory) `__init__()` I do all the checking etc. while the `rtl()` member function handles all the logic. Most of the time I create all output signals in the the `__init__()` section. E.g.:
I should (will someday) write a better document on this.

> 4. Testing in python does not "resolve" std_logic signal in the classical sense, but rather runs the description sequentially, let's say

Obviously there is no `std_logic` attribute in Python. The simulation may run sequentially, as any single thread program, but all *HDL events* are treated concurrently in a cycle-accurate way (with possible/probable delta cycles just like a VHDL simulator does)

The follow-up questions:
> 1. would co-simulating with icarus detect multiple drivers? my assumption would be yes, and I'll soon put it to the test

The conversion of the test-bench would fail early ...

> 2. for the purpose of asserting on assignments to input ports, I see that the TristateSignal type has the desirable feature of not having a "next" attribute, so I imagine that asserting on that type in a https://github.com/block would ensure that one can't assign a value to an input port within an @Always. Can you think of a clean design pattern around this?

I suggest you only use the Tristate signal for its intended purposes. Using this as an input might be possible but in my view will only complicate the code structure?
I refer back to 3. above

I rarely simulate MyHDL generated output VHDL, unless I include this a a component inside a higher-level VHDL module.
So far I haven't used Cosimulation, mainly because this isn't working for VHDL (yet?).

I tried the discourse link on my smartphone and it nicely opened in the browser :)

Regards,
Josy

sideprojectslab, 2024-09-26:
Hi, thanks again for the quick reply.

Ill do some further experiments, maybe I can make a simple wrapper around SignalType that will except when using the "next" attribute, but of course, like you suggest,  whatever solution I may come up with shouldn't be more trouble than it is worth.

Regarding cosimulation, I have no problem simulating in verilog, in fact I think I will even convert to verilog only, and just instantiate the verilog module in the VHDL design (I am already doing that). This should give some degree of confidence that the synthesized code is the same that has been simulated.

The Discourse link opens just fine, but trying to "login with github" or "login with google" (so I can post) seems to fail. Looking forward to join the community over there.

closed 2024-09-26 by sideprojectslab