myhdl/myhdl #443: Accessing/Driving internal signals during simulation

sideprojectslab, 2024-10-03:
Hello,

Is there a way to read/drive internal signals of a DUT during simulation/cosimulation without exposing them as ports? This is a feature that I greatly enjoy in CoCoTB, as it allows to set an module's internal state very easily, in order to test various corner cases without having to drive said state through an input vector.

P.S. Discourse seems pretty barren these days, I posted a couple of questions and I got basically no views in a couple of days. Is Gitter perhaps the place to be?

josyb, 2024-10-04:
Hi,

First about Discourse: there certainly were/are viewers, although nobody felt compelled to answers. As I don't get notifications from  Discourse I check the discussion on an irregular basis and I missed your questions ...
Maybe we better use the 'Discussions' section in this repo? I am almost sure I would get an email if you post there.

Cosimulation: I haven't used this feature - as it is not supported for VHDL.
So I can't help much here. I also have never used CoCoTB so I don't have any idea how this stimulation works.
Do you have  a minimal but complete example I can run?

sideprojectslab, 2024-10-04:
Duly noted, I will use the "discussions" tab for future questions

Regarding those internal signals, before we even start talking about cosimulation, cocotb or any concrete example, let's assume I am simulating in python, so I will have my DUT instantiated in a test, a stimulus generator and a monitor which drive and read the DUT's interfaces respectively, so far so good.

However, take for instance the "blink" module we have seen before, it has an internal counter register. What happens, for instance if I want to read/write the counter register from the monitor/stimulus, so that I can assert on its value or set its value to a specific quantity for the purpose of testing?

One obvious way would be to actively expose the internal signal (counter) as a port for the Blink module, but that doesn't scale well for large architectures as it pollutes the interface with signals that should otherwise be strictly internal. Ideally in a test i'd like to say something like: assert(dut.counter, 100, "Counter reaches 100"), without having to expose counter on a port.

This is just a simple example of course, but such feature is crucial to test corner cases of complex systems.

josyb, 2024-10-04:
Yes, your blink.py is a good thing to start with.
The internal counter signal `count` lives within the _sigdict_ attribute of the `dut`.
Search for the 'MAGIC' in the following code:

```python
from myhdl import Signal, SignalType, always, always_seq, intbv, modbv, block, Constant, instances

class BlinkIO:

    def __init__(self):
        self.enable = Signal(bool(1))
        self.blink = Signal(bool(0))

@block
def blink(clk:Signal(bool), rst:Signal(bool), io:BlinkIO, period: Signal(intbv) | Constant(intbv) | int):

    assert isinstance(clk, SignalType) and(isinstance(clk.val, bool))
    assert isinstance(rst, SignalType) and(isinstance(rst.val, bool))

    count = Signal(modbv(0)[20:])

    @always(clk.posedge)
    def synch():

        if io.enable:
            if count == period:
                count.next = 0
                io.blink.next = not io.blink
                # stop counting (and blinking ...)
                # io.enable.next = 0

            else:
                count.next = count + 1

        if rst:
            count.next = 0

    return synch

if __name__ == '__main__':
    from myhdl import instance, delay, StopSimulation, ResetSignal

    @block
    def tb_blinky():

        clk = Signal(bool(0))
        rst = Signal(bool(1))
        io = BlinkIO()

        dut = blink(clk, rst, io, 200)

        @always(delay(10))
        def clkgen():
            clk.next = not clk

        @instance
        def stimulus():

            yield clk.posedge
            rst.next = 0
            rst2.next = 0
            io.enable.next = 1

            # expect at least one edge
            yield io.blink.posedge
            for __ in range(10 * 100 + 50):
                yield clk.posedge

            # I only put this to easier find the changes in the waveform
            io.enable.next = 0
            yield clk.posedge
            io.enable.next = 1
            yield clk.posedge

            # Here the MAGIC happens
            # let's poke the `count`
            count = dut.sigdict['count']
            print('simulation 1:', count, repr(count))
            # give it a good _kick_
            count.next = count + 100
            # the above will replace or supersede the `count.next = count + 1` value registered at the previous clock edge 
            # we don't see a change here yet as no _update_ event has taken place yet
            print('simulation 2:', count)
            yield clk.posedge
            # the clock event now has updated the value in `count`
            print('simulation 3:', count)

            for __ in range(10):
                yield clk.posedge

            # the `count` should be 10 more now
            print('simulation 4:', count)

            raise StopSimulation

        return instances()

    def test():
        dft = tb_blinky()
        dft.config_sim(trace=True, tracebackup=False)
        dft.run_sim()

```

The simulation output:
```
simulation 1: 0002d Signal(modbv(45))
simulation 2: 0002d
simulation 3: 00091
simulation 4: 0009b

```

sideprojectslab, 2024-10-04:
Oh this is absolutely excellent, thank you so much for taking the time!

closed 2024-10-04 by sideprojectslab