myhdl/myhdl #449: Please switch from setup-ghdl-ci to setup-ghdl

Paebbels, 2025-01-12:
> [!CAUTION]
> This GitHub Action (`setup-ghdl-ci`) has been **deprecated** in favor of the new [setup-ghdl](https://github.com/ghdl/setup-ghdl) Action.
> Please update your YAML files and transition to the new GitHub Action. See the linked repository for details on optional parameters.

**Quick example:**
```yml
jobs:
  GHDL-on-Ubuntu:
    runs-on: ubuntu-24.04
    steps:
      - name: Run VHDL Simulation
        uses: ghdl/setup-ghdl@v1
        with:
          version: nightly
          backend: mcode
```

**Why has it been changed?**  
* The Javascript GitHub Action is hard to maintain.  
  In addition, it has a high complexity with Node.js and other tools compare to a much simpler Composite Action using simple shell instructions in Bash/Powershell.
* This action doesn't support the `ubuntu-24.04` (= `ubuntu-latest`) runner image.
* This Action uses wrong tagging to select GHDL versions. See analysis: https://github.com/ghdl/ghdl/issues/2852
* Due to changes in GHDL's nightly releases and attached assets, the filenames are broken.

-----------------------
This repository has been listed by [Dependabot](https://github.com/ghdl/setup-ghdl-ci/network/dependents) as a dependent of `setup-ghdl-ci`.

davekeeshan, 2025-01-13:
@Paebbels I have made the changes to the CI flow, however the `backend: llvm`, does not appear to be recognised.  myhdl only uses the `setup-ci` from the windows configuration.

https://github.com/myhdl/myhdl/actions/runs/12756629221/job/35555176047?pr=450

[image]

The documentation suggests that this is the correct way:

https://github.com/ghdl/setup-ghdl#input-parameters

Paebbels, 2025-01-14:
@davekeeshan is this your YAML code?
```
      - if: ${{ env.CI_TARGET == 'ghdl' }}
        uses: ghdl/setup-ghdl@v1
        with:
          version: nightly
          backend: llvm
```

I assume you want to run GHDL with LLVM backend within MSYS2, right? MSYS2 is setup (green arrow).  
But `setup-ghdl` sets up GHDL for Windows native, not for MSYS2 (red arrow).

[image]

In case of MSYS2, you also need to specify the `runtime` parameter, otherwise the action cannot decide if to run a powershell (native windows) a bash (git bash in native windows) or a bash inside of MSYS2 with MinGW64.

The documentation lists this example code:
```yaml
  GHDL-on-Windows-with-UCRT64:
    runs-on: windows-2022
    steps:
      - name: 🟦 Setup MSYS2 for UCRT64
        uses: msys2/setup-msys2@v2
        with:
          msystem: ucrt64
          update: true

      - name: Run VHDL Simulation
        uses: ghdl/setup-ghdl@v1
        with:
          version: nightly
          backend: mcode
          runtime: ucrt64
```

UCRT64 is similar to MinGW64. Actually, you should transition from MinGW64 to UCRT64, as this runtime is declining inside of MSYS2.

The `runtime` parameter is documented as follows:
> If runner OS is Windows, a MSYS2 runtime can be selected (mingw64, ucrt64). If not set, Windows native is used (not MSYS2 environment).

Any suggestions on changing the documentation?

The LLVM backend doesn't exist (yet?) for native Windows.

------------------
Btw, you can simplify this code:
```yaml
if: ${{ env.CI_TARGET == 'ghdl' }}
```
to
```yaml
if: matrix.target == 'ghdl'
```
* no need for ${{ ... }}}
* no need for an intermediate global environment variable  
  Please also be aware of several serious issues in GitHub Actions to the order of environment variable evaluations. Basically, you can consider this feature totally broken except for very basic usage models. A rule of thumb tries to avoid environment variables in GitHub's YAML code. (This bug/feature was reported when GitHub Action was still in testing phase several years ago).

davekeeshan, 2025-01-14:
@Paebbels Thanks for that, I needed to make sure path-type was also included, without it actions could not find python.

      - uses: msys2/setup-msys2@v2
        with:
          msystem: ucrt64
          path-type: inherit
          update: true

Thanks for the improvement notes, I think some might have been there from a legacy perspective, but it is good to have a fresh set of eyes every few years.

I have also moved the whole flow, including the linux part under `ghdl/setup-ghdl` the quality of the ghdl actions nightly build seems to be up to par now. (I wish there was one for icarus too)

Paebbels, 2025-01-15:
@davekeeshan I think your MSYS2 is not setup correctly.

> I needed to make sure path-type was also included, without it actions could not find python.

This means you're using the Python from Windows, not the Python within MSYS2/UCRT64. While this is technically possible (as you achieved it), it's not recommended nor does your test environment behave like other user's environments look like. For a "correct" setup, install Python within MSYS2.

```yaml
  - uses: msys2/setup-msys2@v2
    with:
      msystem: ucrt64
      update: true
      install: git                             # MSYS2 packages like git, make, ...
      pacboy: python-pip:p python-wheel:p      # runtime specific packages like Python
```

`pacboy` is a helper program like `pacman`. I uses `<name>:p` for packages and automatically resolves the package name to a mingw64 or ucrt64 package name.
In case you need normal programs, I also added how to install `git`. If not needed, please remove that with-parameter.

---------------------
> I have also moved the whole flow, including the linux part under ghdl/setup-ghdl the quality of the ghdl actions nightly build seems to be up to par now. 

All builds on GHDL's main branch is very stable. We move the nightly tag from time to time. So it should be as stable as the main builds. In case we get problems reported, Tristan usually pushes a fix within days to the main branch or we could revert the nightly tag.

davekeeshan, 2025-01-15:
> @davekeeshan I think your MSYS2 is not setup correctly.
>
>
> This means you're using the Python from Windows, not the Python within MSYS2/UCRT64. While this is technically possible (as you achieved [...]

That seems messy and confusing, is the `actions/setup-python` actually being used in that case and how about setting the version of python, I have not used in it matrix mode proper here, but was considering adding one or two more, this seems decouple from that.

Paebbels, 2025-01-15:
> That seems messy and confusing, is the `actions/setup-python` actually being used in that case [...]

Nope. `actions/setup-python` has no real effect. Yes it installs Python in native Windows, but this isn't used by MSYS2. So you can deactivate it in case you matrix uses UCRT64. This saves a few seconds of job time.

> [...] and how about setting the version of python, [...]

MSYS2 uses pacman and rolling releases. So MSYS2 always uses latest versions. Thus, you can't really select the Python version. Currently, it's 3.12, because they have some problems porting 3.13 to MSYS2 environment. This again is due to egoists at Python Packaging Authority and other parts of Python refusing "any" support of MSYS2. So the MSYS2 team has to manually port and patch CPython. And non of the patches are going upstream, nor are breaking modifications communicated to MSYS2 ...

-------
You can checkout my YAML code for inspiration how to deal with such problems:
1) There's a collection of workflow templates:  
   https://github.com/pyTooling/Actions/main/dev/.github/workflows  
2) There is the UnitTesting.yml workflow template:  
   https://github.com/pyTooling/Actions/blob/main/.github/workflows/UnitTesting.yml  
3) The matrix is written by Python code as JSON output:  
   https://github.com/pyTooling/Actions/blob/main/.github/workflows/Parameters.yml  
4) Putting all together:  
   https://github.com/pyTooling/Actions/blob/main/.github/workflows/CompletePipeline.yml  
5) What the user sees:  
   https://github.com/pyTooling/Actions/blob/main/.github/workflows/_Checking_SimplePackage_Pipeline.yml

These workflow templates can be used by any pipeline. these are made for Python projects utilizing unit tests, documentation (HTML, LaTeX), matrix builds, code coverage, result publishing, ...