myhdl/myhdl #452: traceSignals arguments doesn't seem to work

cychotic, 2025-02-18:
Summary

## System information
```
MyHDL Version: 0.11.51
Python Version: 3.12.1
```

traceSignals arguments doesn't seem to work.  The arguments 'directory' and 'filename' have no effect and it always generate the default output filenames.

Example code
```
    tb = testbench_axis_to_serial()

    # Ensure the directory exists
    vcd_dir = "vcd"
    if not os.path.exists(vcd_dir):
        os.makedirs(vcd_dir)

    # Use traceSignals with directory and filename arguments
    tb_with_trace = traceSignals(tb, name="test", directory=vcd_dir, filename="axis_to_serial", timescale='10ns')

    # Run simulation
    sim = Simulation(tb_with_trace)
    sim.run()
    sim.quit()
```

cychotic, 2025-02-18:
Temporary workaround

```
    traceSignals.directory = vcd_dir
    traceSignals.filename = "axis_to_serial"
    traceSignals.timescale = "10ns"
    traceSignals.name = "test"
```

josyb, 2025-02-18:
Hi,

I wished that you had showed a little bit more of your code.
First, let's assume that you are not using the `@block` decorator in your code. Or do you?
In which case your workaround is the actual correct way.
The documentation however describes what you tried first ...
At some point in time we will *remove* the *old* way and force the use of the `@block` decorator

In the other case that you have used the `@block` decorator (which I now think you do):
```python
@block
def  testbench_axis_to_serial():
     . . . 

dft = testbench_axis_to_serial()
dft.config_sim(trace=True, tracebackup=False,  name="test", directory=vcd_dir, filename="axis_to_serial", timescale='10ns')
dft.run_sim()
```
should do what you expect.

cychotic, 2025-02-18:
Hi josyb, I am using @block decorator.  I just tried config_sim and it seems to work, but using traceSignals directly still doesn't work.  Is config_sim the preferred way to run simulation?

josyb, 2025-02-19:
Yes, `config_sim` and `run_sim` are the preferred way, and soon the only way ...

closed 2025-03-31 by cychotic