myhdl/myhdl #453: Shadow Signal doesn't generate correct Verilog

cychotic, 2025-03-31:
Summary

For the verilog output, shadow_data_in is always set to 0 instead of correctly bit slicing the data_in signals.  VHDL seems to be correct.  I wrote a demo code to demonstrate this.

## System information
```
MyHDL Version:  0.11.51
Python Version:  3.12.1
```

MyHDL Code:

```
from myhdl import block, Signal, intbv, always_seq, always_comb, ResetSignal, ConcatSignal

@block
def shadow_demo(
    clk_i,          # Clock input
    reset_i,        # Reset input
    data_in,        # variable-bit input data
    data_out,       # variable-array output of 8-bit data 
    BYTE_WIDTH,
    DATA_WIDTH,
):
    # Define constants
    NUM_BYTES = DATA_WIDTH // BYTE_WIDTH
       
    # Create shadow signals for each byte
    shadow_data_in = [data_in((NUM_BYTES - 1 - x) * 8 + 8 , (NUM_BYTES - 1 - x) * 8) for x in range(NUM_BYTES)]

    internal_word = [Signal(intbv(0)[BYTE_WIDTH:]) for _ in range(NUM_BYTES)]
    shadow_internal_word = ConcatSignal(*[shadow_data_in[x] for x in range(NUM_BYTES)])

    @always_comb
    def comb():
        for x in range(NUM_BYTES):
            data_out[(NUM_BYTES - 1 - x) * 8 + 8 : (NUM_BYTES - 1 - x) * 8].next = shadow_internal_word[x]

    @always_seq(clk_i.posedge, reset=reset_i)
    def seq():
        for x in range(NUM_BYTES):
            internal_word[x].next = shadow_data_in[x]
    
    return comb, seq

# Conversion function
def convert():

    DATA_WIDTH = 32
    BYTE_WIDTH = 8

    # Define signals
    clk_i = Signal(bool(0))
    reset_i = ResetSignal(0, active=1, isasync=True)
    data_in = Signal(intbv(0)[DATA_WIDTH:])
    data_out = Signal(intbv(0)[DATA_WIDTH:])

    
    # Create instance
    inst = shadow_demo(clk_i, reset_i, data_in, data_out, BYTE_WIDTH, DATA_WIDTH)
    
    # Convert to Verilog
    inst.convert('Verilog', name='shadow_demo')
    
    # Convert to VHDL
    inst.convert('VHDL', name='shadow_demo')

if __name__ == '__main__':
    convert()
```

Verilog Output:

```
// File: shadow_demo.v
// Generated by MyHDL 0.11.51
// Date:    Mon Mar 31 16:07:07 2025 UTC

`timescale 1ns/10ps

module shadow_demo (
    clk_i,
    reset_i,
    data_in,
    data_out
);

input clk_i;
input reset_i;
input [31:0] data_in;
output [31:0] data_out;
reg [31:0] data_out;

wire [31:0] shadow_internal_word;
reg [7:0] internal_word [0:4-1];
reg [7:0]  shadow_data_in [0:4-1];
initial begin
    shadow_data_in[0] <= 0;
    shadow_data_in[1] <= 0;
    shadow_data_in[2] <= 0;
    shadow_data_in[3] <= 0;
end

assign shadow_internal_word[32-1:24] = shadow_data_in[0];
assign shadow_internal_word[24-1:16] = shadow_data_in[1];
assign shadow_internal_word[16-1:8] = shadow_data_in[2];
assign shadow_internal_word[8-1:0] = shadow_data_in[3];

always @(shadow_internal_word) begin: comb
    integer x;
    for (x=0; x<4; x=x+1) begin
        data_out[((((4 - 1) - x) * 8) + 8)-1:(((4 - 1) - x) * 8)] = shadow_internal_word[x];
    end
end

always @(posedge clk_i, posedge reset_i) begin: seq
    integer x;
    if (reset_i == 1) begin
        internal_word[0] <= 0;
        internal_word[1] <= 0;
        internal_word[2] <= 0;
        internal_word[3] <= 0;
    end
    else begin
        for (x=0; x<4; x=x+1) begin
            internal_word[x] <= shadow_data_in[x];
        end
    end
end

endmodule

```

VHDL Seems to be correct:
```
-- File: shadow_demo.vhd
-- Generated by MyHDL 0.11.51
-- Date:    Mon Mar 31 16:07:07 2025 UTC

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_011.all;

entity shadow_demo is
    port (
        clk_i: in std_logic;
        reset_i: in std_logic;
        data_in: in unsigned(31 downto 0);
        data_out: out unsigned(31 downto 0)
    );
end entity shadow_demo;

architecture MyHDL of shadow_demo is

signal shadow_internal_word: unsigned(31 downto 0);
type t_array_internal_word is array(0 to 4-1) of unsigned(7 downto 0);
signal internal_word: t_array_internal_word;
type t_array_shadow_data_in is array(0 to 4-1) of unsigned(7 downto 0);
signal shadow_data_in: t_array_shadow_data_in;

begin

shadow_internal_word(32-1 downto 24) <= shadow_data_in(0);
shadow_internal_word(24-1 downto 16) <= shadow_data_in(1);
shadow_internal_word(16-1 downto 8) <= shadow_data_in(2);
shadow_internal_word(8-1 downto 0) <= shadow_data_in(3);
shadow_data_in(0) <= data_in(32-1 downto 24);
shadow_data_in(1) <= data_in(24-1 downto 16);
shadow_data_in(2) <= data_in(16-1 downto 8);
shadow_data_in(3) <= data_in(8-1 downto 0);

comb: process (shadow_internal_word) is
begin
    for x in 0 to 4-1 loop
        data_out(((((4 - 1) - x) * 8) + 8)-1 downto (((4 - 1) - x) * 8)) <= to_unsigned(shadow_internal_word(x), 32);
    end loop;
end process comb;

seq: process (clk_i, reset_i) is
begin
    if (reset_i = '1') then
        internal_word(0) <= to_unsigned(0, 8);
        internal_word(1) <= to_unsigned(0, 8);
        internal_word(2) <= to_unsigned(0, 8);
        internal_word(3) <= to_unsigned(0, 8);
    elsif rising_edge(clk_i) then
        for x in 0 to 4-1 loop
            internal_word(x) <= shadow_data_in(x);
        end loop;
    end if;
end process seq;

end architecture MyHDL;

```

josyb, 2025-04-02:
Yes, there is an issue with generating Verilog for your code.
This is my a work-around:
```python
@block
def shadow_demo(
    clk_i,  # Clock input
    reset_i,  # Reset input
    data_in,  # variable-bit input data
    data_out,  # variable-array output of 8-bit data
    BYTE_WIDTH,
    DATA_WIDTH,
    ):
    # Define constants
    NUM_BYTES = DATA_WIDTH // BYTE_WIDTH

    # Create shadow signals for each byte

    # work-around
    # shadow_data_in = [data_in((NUM_BYTES - 1 - x) * 8 + 8 , (NUM_BYTES - 1 - x) * 8) for x in range(NUM_BYTES)]
    shadow_data_in = [Signal(intbv(0)[BYTE_WIDTH:]) for __ in range(NUM_BYTES)]

    internal_word = [Signal(intbv(0)[BYTE_WIDTH:]) for __ in range(NUM_BYTES)]
    # no need to re-create a ListOfSignals to feed it to ConcatSignal
    # shadow_internal_word = ConcatSignal(*[shadow_data_in[x] for x in range(NUM_BYTES)])
    shadow_internal_word = ConcatSignal(*shadow_data_in)

    @always_comb
    def comb():
        for x in range(NUM_BYTES):
            # correcting a .next placement
            # Note: conversion does handle it, bu simulation won't
            # and one really should simulate first before attempting conversion
            # data_out[(NUM_BYTES - 1 - x) * 8 + 8: (NUM_BYTES - 1 - x) * 8].next = shadow_internal_word[(x + 1) * 8:x * 8]
            data_out.next[(NUM_BYTES - 1 - x) * 8 + 8: (NUM_BYTES - 1 - x) * 8] = shadow_internal_word[(x + 1) * 8:x * 8]

            # work-around
            shadow_data_in[x].next = data_in[(NUM_BYTES - 1 - x) * 8 + 8: (NUM_BYTES - 1 - x) * 8]

    @always_seq(clk_i.posedge, reset=reset_i)
    def seq():
        for x in range(NUM_BYTES):
            # Note: internal_word is never read ... perhaps MyHDL should display a warning
            # I suppose this code is for demonstration only?
            internal_word[x].next = shadow_data_in[x]

    return comb, seq
```

ShadowSignals were primarily created to pass them as arguments in an instantiation of another module, e.g.:
```python
    modinst = some_module(Clk, ..., data_in(16,8), ...)
```
Your usage was not envisioned (I guess), as there is no test written for it. But apparently someone has created the code for correct VHDL conversion ...

cychotic, 2025-04-02:
Thank you so much for the response.

The workaround you provide is actually why I tested a potential use case of the Shadow Signals.

I needed a variable width of data will need to be byte accessed.  So in fact, I actually started with something very similar to this.
```
data_out.next[(NUM_BYTES - 1 - x) * 8 + 8: (NUM_BYTES - 1 - x) * 8] = shadow_internal_word[(x + 1) * 8:x * 8]
```

However, with Verilog, I would get the famous "not a constant" error for slicing with a variable  `[(x + 1) * 8:x * 8]`.

josyb, 2025-04-24:
You can start by reading these two:
https://dev.myhdl.org/guide/guide.html
https://dev.myhdl.org/guide/guide_tests.html